Opentrons/opentrons #637: App Split LabwareList into TipwareList and LabwareList

Kadee80, 2018-01-19:
## Overview
Currently in the App all labware is reduced within the `LabwareList` container and renders a `TitledList`  for all `LabwareItems` with the grouped tipracks first.

New `SetupPanel` sidebar designs call for the separation of `LabwareList` into 2  titled lists,  1 for tipracks and 1 for all other labware. 

These lists will need to allow for disabling of individual `TipRackListItems` as well as disabling entire `TitledLists`. 
- For MVP, once a tiprack is calibrated, is can not be revisited. (disable individual item)
- Labware  titled list is disabled until all tipracks are confirmed.
- Once all tipracks confirmed, tipracks titled list is disabled.

These 2 TitledLists will need to be grouped and titled via an upcoming new comp lib component called `SidbarGroup` (separate ticket).

[image]

## Steps to implement
- [ ] Render tipracks and other labware as 2 different titled lists.
- [ ] Update TitledList to include `isDisabled` prop. 
- [ ] Update styles to match designs (headings in particular
- [ ] Wrap labware sidebar section in `SidebarGroup` (this is blocked until `SidebarGroup` exists.)

Kadee80, 2018-01-19:
@pantslakz the current app has some alert/warning text explaining that:
- labware is disabled until pipettes are probed 
- other labware is disabled until tipracks are calibrated. 

This disabling of entire groups, lists, or list items seems pretty clear. Are we removing these alerts?

pantslakz, 2018-01-22:
@Kadee80 yes we can remove these. In place of that text, we now have the hover tip as shown below:

(to show when user hovers over greyed out options)
[image]

PR #716 (2018-02-05): App split tiprack labware lists
## overview
This is a WIP PR that should close #637. Runs as expected locally but implementing flow in `TiprackList` and `LabwareList` caused errors in untouched files. (As well as multiple errors in those files that seem to change unexpectedly in atom)

## changelog

- Implement SidePanelGroup for Calibrate Deck section
- Split LabwareList into TiprackList and LabwareList containers
- Implement logic for disabling entire group, individual lists, or individual list items.
- Implement flow, (probably caused more problems fixing errors than I think)

## review req [... cut]
closed 2018-02-05 by Kadee80