Opentrons/opentrons #645: CSS Fixes for Deck and Plate component

IanLondon, 2018-01-19:
## overview

This issue is a stub, reminding me to fully ticket out the problems I commented in on PR #633 

- [ ] Plate label font size
- [ ] Deck `#roundSlotClipPath` no longer works for PD's Add/Copy/Remove overlays
- [ ] Hovering over labware name overlay mysteriously indefinitely hides the overlay even when you mouse off it

mcous, 2018-02-09:
We should add viewbox corrections to this list, too

IanLondon, 2018-07-24:
@mcous are there newer tickets for this?

mcous, 2018-07-25:
@IanLondon don't think so! The font-size and clip-path stuff is still in need of work I think (dunno about the hover stuff). We can keep this one around or close it in favor of a new refactor ticket when we have one

IanLondon, 2019-01-11:
@b-cooper is working on deck refactor, this ticket is out of date

closed 2019-01-11 by IanLondon