Opentrons/opentrons #648: Update Jobs Board to Google Hire

umbhau, 2018-01-22:
## Overview
As a potential job applicant, I'd like to be able to see and apply to job posted through Google Hire, rather than Recruitee.

## Acceptance Criteria
- User able to view a new job page with job postings from Google Hire (new page with Google Hire embed code, below)
- User able to visit page from all "we're hiring" links (update all links to point to new page)
- User able to submit job applications to Google Hire platform

## Resources
- Embed Code:
<div class="bebop-jobs"></div>
<script id="bebop-embed-loader" async defer src="https://hire.withgoogle.com/s/embed/jobs.js?company=opentronscom"></script>

Kadee80, 2018-01-23:
Do to the fact that our website is built in view, we can't just inject the script tag on the page easily. The workaround will be to link to the published jobs page hosted by google hire:
`https://hire.withgoogle.com/public/jobs/opentronscom` OR try embedding that page within the site. 

We currently have no published open jobs, so in addition to changing the links (will discuss in sprint planning) we need to publish new jobs before making this live.

Kadee80, 2018-01-23:
[staging server for discussion/ review](http://website_add-google-hire.facelift.opentrons.com/jobs)

Kadee80, 2018-01-24:
Closed via https://github.com/Opentrons/opentrons-dot-com/pull/72

closed 2018-01-24 by Kadee80