Opentrons/opentrons #650: Pipette slow aspirate rate

jacobfeder, 2018-01-22:
## overview

 Trying to aspirate at less than 60mm/minute.

## behavior

Setting aspirate rate to anything under 60mm/sec results in the machine stalling when it should be moving. This also appears to make it lose track of where it is.

Is there a (int)(rate/60) conversion from mm/minute to mm/sec getting rounded down to 0 somewhere?

mcous, 2018-07-12:
Hey @jacobfeder, I'm really sorry this wasn't followed up on! I know it's been a while, but if this is still an issue you care about, would you mind providing the following information (or as much as you have / remember)?

- Robot model (e.g. OT-1 or OT-2)
- Software version
- OS and version
- Protocol if available / shareable

jacobfeder, 2018-07-12:
It's an OT-1 not sure about software version. I think you will be able to repeat it by making a simple protocol that sets the aspirate rate to e.g. 30mm/sec. If not I can try to provide more info.

mcous, 2018-07-13:
Thanks, I appreciate the quick feedback! I was unable to find any obvious integer math problems with speed setting; the [motor driver board](http://smoothieware.org/smoothieboard) in the OT-1 deals directly with integers in mm/minute, so the software does almost no math before passing along rates. The OT-1 version of the software lives over in the [`ot1-stable`](https://github.com/Opentrons/opentrons/tree/ot1-stable) branch if you're curious to check it out.

From your description (motors stalling out) this sounds like it may be a fundamental hardware limitation. Unfortunately, since we're a relatively small team, our engineering resources are 100% focused on the OT-2 rather than the OT-1. I'm going to close this issue out because it's very unlikely we issue any OT-1 fix, even if there is a bug somewhere.

That being said, I'll definitely ask around with our hardware team to see if there may be a workaround and update this thread accordingly.

closed 2018-07-13 by mcous