Opentrons/opentrons #651: Asserition error from calibration_manager.tip_probe after labware calibration

mcous, 2018-01-23:
## overview

When going through calibration, if a user...

1. Completes tip probe
2. Completes tiprack calibration
3. Goes back to tip probe for the labware calibrator pipette

...the API throws an `AssertionError` when the client tries to call `calibration_manager.tip_probe`.

This blocks #623 from being fully completed, but doesn't block the implementation of the UI.

## behavior

### expected

The user should be able to restart tip-probe at any point prior to a run, and they are (or rather, will be at the conclusion of #623) given clear instructions on when to add and remove tips by hand.

### actual

The API's attempts at blocking multiple tip-pickups prevent the user from re-probing the tip used for labware calibration.

## steps to reproduce

1. Open app in debug mode (with console open) with virtual-robot API
2. Load a protocol (I used `bradford_assay.py` from `data`)
3. Go through tip probe
4. Calibrate all tip racks
5. Click on the calibrator pipette (i.e. the single channel pipette) and click "Start Tip Probe"
6. Click "Continue"
7. Notice the `robot:PROBE_TIP_RESPONSE` come back with `error: true` and the API log the following trace:

```
2018-01-22 19:11:56,177 opentrons.server.rpc WARNING [Line 276] Exception while dispatching a method call: Traceback (most recent call last):
  File "/Users/mc/opentrons/opentrons/api/opentrons/server/rpc.py", line 271, in make_call
    self.executor, self.call_and_serialize, func)
  File "/Users/mc/.pyenv/versions/3.5.3/Python.framework/Versions/3.5/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/Users/mc/.pyenv/versions/3.5.3/Python.framework/Versions/3.5/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/Users/mc/.pyenv/versions/3.5.3/Python.framework/Versions/3.5/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/Users/mc/.pyenv/versions/3.5.3/Python.framework/Versions/3.5/lib/python3.5/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/mc/opentrons/opentrons/api/opentrons/server/rpc.py", line 261, in call_and_serialize
    call_result = func()
  File "/Users/mc/opentrons/opentrons/api/opentrons/api/calibration.py", line 39, in tip_probe
    robot=inst.robot)
  File "/Users/mc/opentrons/opentrons/api/opentrons/util/calibration_functions.py", line 55, in probe_instrument
    instrument._add_tip(tip_length)
  File "/Users/mc/opentrons/opentrons/api/opentrons/instruments/pipette.py", line 1596, in _add_tip
    assert not self.tip_attached
AssertionError
```

btmorr, 2018-01-24:
This will be resolved by #536. Tip probe requires no tip on at start. This error happens because we currently leave the tip on at the end of calibration. After #536 we will no longer leave the tip on at the end of calibration, so this assertion will no longer be thrown

mcous, 2018-01-30:
Closing in favor of #536

closed 2018-01-30 by mcous