Opentrons/opentrons #653: Support multiple trash containers for a pipette

antonjs, 2018-01-23:
## overview

It would be great if you could specify multiple trash containers for a pipette, and/or multiple locations within a trash container where tips can be dropped. Even better, the pipette would randomly choose a location within a container to drop tips, rather than going to the same point every time.

## behavior

During long protocols, tips build up in a pile in the trash container, under the point that defines the trash. Eventually tips can't be dropped properly because they start hitting the pile. When using a single channel, sometimes a tip being dropped will actually spear an old tip that is standing upright at the same point. 

Spreading tips out throughout the container would help solve this problem, as would the ability to use multiple trash containers automatically. Multiple containers would also allow multiple points to be calibrated within the same tip box.

## steps to reproduce

Use a box of tips with one container.

Matt-Zwimpfer, 2021-06-03:
MMM - research into API for this

alitru6, 2023-03-07:
> 

hello! have there been any updates to programming the robot to drop tips in a random/different location in the fixed trash? we ran into the same problem using multichannel tips that didn't fit after 2 racks because they were being dropped in the same location
OT-2 running Zymo Zyppy™-96 Plasmid MagBead Miniprep protocol

SyntaxColoring, 2024-01-02:
As of robot software v7.0, robots can automatically drop tips in alternating locations within the trash bin.

If you have a Python protocol, make sure its `apiLevel` is at least `2.15`. See [the Python Protocol API version notes](https://docs.opentrons.com/v2/versioning.html#version-2-15). If you have a Protocol Designer protocol, you might need to import it into Protocol Designer and then re-export it to opt in to the new behavior—not sure.

closed 2024-01-02 by SyntaxColoring