Opentrons/opentrons #660: Multichannel does not dispense or pick up tips correctly

Laura-Danielle, 2018-01-23:
## overview

During tip probe as well as protocol workflows, the multichannel pipette does not pick up tips as expected. This causes the tips to not have a good seal on the pipette and thus volumes are inconsistent.

## behavior
The pipette will move to the correct location in the container, move very slowly/gently down to the correct position, move up another ~10mm and make the "pick up tip" motion in the air. **This happens for all types of containers.**

## steps to reproduce

1. Load in a multichannel pipette to your protocol
2. Perform tip probe
3. Attempt to calibrate the containers
4. Run the protocol with the calibrated containers (where you will see the "pick up tip" motion for all containers).

andySigler, 2018-01-28:
Unable to reproduce on latest `v3a`, running `bradford_protocol.py` the multichannel performs `pick_up_tip()` correctly. Are you able to share the protocol that was causing this behavior, and also which machine it was on?

andySigler, 2018-01-29:
Running below scripts (which was reported to have reproduced this error) does not cause any strange behavior on `v3a` commit `dfd977e`:

```python
"""
@author opentrons
@date January 18th, 2018
@version V1.2
"""
from opentrons import containers, instruments, robot

VOLUME_ONE = 150
VOLUME_TWO = 9
VOLUME_THREE = 8
VOLUME_FOUR = 7
VOLUME_FIVE = 6
# labware definitions
tiprack1 = containers.load('tiprack-200ul', 'C1', 'Tiprack 1')
tiprack2 = containers.load('tiprack-200ul', 'C2', 'Tiprack 2')
dye_trough = containers.load('trough-12row', 'C3', 'Dye Trough')
trash = containers.load('trash-box', 'C4')

"""
Below are 8 separate 96 well plates that can be processed.
They are separated out into half full plates and full plates.
If you wish to run all of the plates as one setting see below.
The numbers next to the plate represents the slot number
"""
plate_2 = containers.load('96-flat', 'B1')
plate_5 = containers.load('96-flat', 'B2')
plate_8 = containers.load('96-flat', 'B3')
plate_11 = containers.load('96-flat', 'B4')

plate_1 = containers.load('96-flat', 'A1')
plate_4 = containers.load('96-flat', 'A2')
plate_7 = containers.load('96-flat', 'A3')
plate_10 = containers.load('96-flat', 'A4')

"""
All four pipettes are loaded into the program, however the app will only
see pipettes that are currently being used.
"""

p300multi = instruments.Pipette(
    mount='left',
    name="p300 Multi",
    tip_racks=[tiprack1, tiprack2],
    trash_container=trash,
    channels=8)

plates = [
    plate_1,
    plate_2,
    plate_4,
    plate_5,
    plate_7,
    plate_8,
    plate_10,
    plate_11]

"""
Variable initialization
@pip set the current pipette to the name being tested; i.e. if you want to test
the 10ul multichannel, then you would set pip = p10multi
@num_plates is the amount of plates you are currently processing -
we can play around with how this is implemented after you use the protocol :)
"""

pip = p300multi
num_plates = 1

# make sure this matches with the amount of plates you want to process

"""
Helper functions
"""

def pre_wet(well):
    pip.aspirate(50, well)
    pip.dispense(50, well)

def add_dye(volume, well):
    pip.pick_up_tip(presses=1)
    pre_wet(dye_trough[0])
    pip.aspirate(volume, dye_trough[0])
    pip.dispense(volume, well.bottom())
    pip.blow_out()
    pip.return_tip()

"""
Creates a list of plate configurations
so that you don't have to keep writing the
same code over again.
"""
# wells to add dye to
dye_test_one = []
dye_test_two = []
dye_test_three = []
dye_test_four = []
dye_test_five = []

# Plate 1 configuration
dye_test_one.append(plate_1.rows(0)[4:])
dye_test_two.append([])
dye_test_three.append([])
dye_test_four.append([])
dye_test_five.append([])

# Plate 2 configuration (if you want not put a die in the plate, simply
# add an empty list)

for val in range(num_plates):
    if val == 1:
        VOLUME_ONE = 5
        VOLUME_TWO = 4
        VOLUME_THREE = 3
        VOLUME_FOUR = 2
        VOLUME_FIVE = 1
    for well in dye_test_one[val]:
        add_dye(VOLUME_ONE, well)
    for well in dye_test_two[val]:
        add_dye(VOLUME_TWO, well)
    for well in dye_test_three[val]:
        add_dye(VOLUME_THREE, well)
    for well in dye_test_four[val]:
        add_dye(VOLUME_FOUR, well)
    for well in dye_test_five[val]:
        add_dye(VOLUME_FIVE, well)

```

andySigler, 2018-01-29:
Update, problem from original script was isolated to pipette's being added to the same mount:
```python
# put a pipette on the LEFT
p1 = instruments.Pipette(mount='left')
# overwrite the previous pipette with a new one on the LEFT
p2= instruments.Pipette(mount='left')
# use the previously overwritten pipette
p1.aspirate()
```
The above code does not raise an error, but when running exhibits very strange movements. Changing it so that instead `p2= instruments.Pipette(mount='right')` so no pipettes are overwritten removes the movement bug.

PR #688 (2018-01-29): Error if pipette mount already occupied

## overview

A bug was found (fixes #660 ) which results in pipettes moving strangely. To recreate, you create a protocol which uses a pipette that has been overwritten by the robot.

```python
p1 = instruments.Pipette(mount='left')
p2= instruments.Pipette(mount='left')
p1.pick_up_tip().aspirate().dispense().return_tip()
```
SOLUTION: Raise a `RuntimeError` if a pipette is attempting to attach to an already occupied mount, resulting in the following:

```python

p1 = instruments.Pipette(mount='left')
# protocol execution stops during below line, because [... cut]
closed 2018-01-29 by btmorr