Opentrons/opentrons #665: The "fast" and "slow" jog controls are confusing on the app

Laura-Danielle, 2018-01-24:
## overview

It is sometimes not clear which setting you are on when jogging the robot.

## behavior

If I double click on the "fast" or "slow" button twice accidentally, it will switch to the opposite setting. This can be confusing as the current graphics are not super clear as to which one is "pressed" and which one is not pressed.

## steps to reproduce

Use the jogger tool, and double click on either setting to see it switch.

mcous, 2018-01-24:
@pantslakz based on slack I'm pretty sure this is already on your radar. This was a engineer special from yours truly. I should note that there's absolutely no technical meaning behind slow and fast, we just picked arbitrary jog distances of 0.25 and 4 mm, respectively. There's also no technical reason to have only two options.

I'm going to remove the "bug" label simply because there's no bug in the code (code is written to toggle distance on on control click, with the non-faded option being the selected on). Perhaps we need a label for bad UX?

pantslakz, 2018-01-24:
design label should do :P

pantslakz, 2018-01-25:
@Laura-Danielle how do you feel about removing the "fast/slow" buttons, and in place enabling user to jog in larger increments when holding [SHIFT] + arrow.

Laura-Danielle, 2018-02-05:
Sorry @pantslakz just seeing this comment for some reason.  Would prefer for the user to just be able to use arrows to jog instead of holding [SHIFT] + arrow. Would also be nice to have more fine-tuning ability on increment changes.

mcous, 2018-04-26:
Closed by #1223

closed 2018-04-26 by mcous