Opentrons/opentrons #692: Update Labware File Format

umbhau, 2018-01-30:
As an Opentrons developer, I’d like to update our labware (a.k.a. 'container') file format so that they're easier to manage.

## Acceptance Criteria

Update labware ‘container’ file format as follows:
- Break labware definitions into individual files (so that we don’t have to manage one huge file)
- Save labware offsets/adjustments from calibration in a separate user-specific json file (so that we don’t overwrite default container positions)
-- Note this implies that to share a protocol user has to share this adjustments file
- Change wells from a dict to a list of lists (so that we can pass the API ordered wells)
-- Implementation note: Include a dictionary as an index?
- Collapse our two offsets (the ‘origin offset’ is for slot A1; each well also has an xyz offset) into one offset (to get rid of redundancy)
- Move labware json definitions into a place that works best for JS apps, and make Docker file copy it into the container (so that we don’t have to use relative paths to import JSON files into JS)
- Replace all instances of the ‘container’ with ‘labware’ (because ‘container’ is super ambiguous at Opentrons)

andySigler, 2018-01-30:
https://github.com/Opentrons/opentrons/issues/700

btmorr, 2018-02-28:
'labware' is now available as an ailas for 'container' everywhere that is relevant in the api. Closing this in favor of specific tickets for future work.

closed 2018-02-28 by btmorr