Opentrons/opentrons #699: Calibrate Robot to Pipette (DRAFT)

umbhau, 2018-01-30:
As a robot user, I would like my robot to automatically calibrate to whichever pipette I am using.

Acceptance Criteria
- Store y-calibration (and tbd other calibration) on robot
- Automatically set calibration when pipette is attached.

Resources
- Alfie/Carlos to provide a spreadsheet of calibration settings

Implementation Notes
- We will eventually move pipette calibration information to pipette memory
- Robots will have to check both pipette (first priority) and internal config file (second priority) for relevant calibration data.

andySigler, 2018-01-30:
Explicitly instantiated pipettes by their model name could look like:
```python
p10 = instruments.P10_Single(mount='left')
m300 = instruments.P300_Multi(mount='right')
```

andySigler, 2018-02-05:
Also fixes #561

PR #745 (2018-02-07): Instantiate pre-configured Pipettes by model

## overview

Fixes #561 and fixes #699 

This PR allows the instantiation of pre-configured pipettes, named by model.

```python
p300_single = instruments.P300_Single(mount='left')
p10_multi = instruments.P10_Multi(mount='right')
```

## changelog

Pipette models are created with `InstrumentsWrapper`, where a `Pipette` is created and the appropriate properties are set:

- pick-up-tip current
- plunger positions (aka top, bottom, blow-out, drop-tip)
- number of channels
- ul-per-mm conversion

To accomodate pipette's with different pick-up-tip curre [... cut]
closed 2018-02-07 by andySigler