Opentrons/opentrons #702: Calibration: tip probe - failure to attach tip correctly

pantslakz, 2018-01-30:
User (Kaja) unable to return to "calibrate tip" screen upon tip probe failure. App did not recognize that pipette had no tip attached during tip probe.

[image]

Ideal behavior:
- once tip probe begins, if no tip attached - user is returned to this screen:
[image]

umbhau, 2018-02-04:
@pantslakz - this solution only works if the tip falls off before the tip probe, right?
"once tip probe begins, if no tip detected - user is returned to this screen:"

but we could solve it generically by building the calibration pause/cancel feature we've been talking about?

pantslakz, 2018-02-04:
@umbhau chatting with Mike what I understood is that we have ability to detect if there's a tip missing at any point during the tip probing process, and if that happens we can bring them back to this screen.

I havent yet heard about a calibration pause/cancel feature, but sounds very useful (especially in case of potential harm to pipette)

mcous, 2018-04-26:
As discussed today, reclassifying this from a "bug" to "designs needed". The app captures errors during tip probe but has nowhere to display them

mcous, 2018-04-26:
Actually just going to close this in favor of #1157

closed 2018-04-26 by mcous