Opentrons/opentrons #705: Pipette with no assigned trash drops tip in the air instead of returning tip to tiprack

IanLondon, 2018-01-31:
## overview

When a pipette is not assigned to a trash container, it `drop_tip` should return the tip to the last empty space in the tiprack. This behavior existed in v2 but has dropped out in v3.

This is mostly useful for demos and tests.

## behavior

Currently, when you explicitly assign no trash to a pipette with `trash_container=None` like this and call drop_tip:

```
from opentrons import robot, containers, instruments
robot.connect()
tr = containers.load('tiprack-200ul', '2')
p = instruments.Pipette(name='p300', mount='right', tip_racks=[tr], trash_container=None)
p.pick_up_tip()
p.drop_tip()
```

...the robot will drop the tip from wherever the pipette is at without moving to the tiprack. (Internally, it's because `drop_tip`'s `location` arg is called with `None`)

## steps to reproduce

The code snippet above should return the tip to the well in the tiprack that it was picked up from.

Laura-Danielle, 2018-02-05:
How recently did you see this behavior @IanLondon ? Was it after the fixed-trash as default PR was merged?

IanLondon, 2018-07-12:
Still true in a30912ff5519af645e364b5db1c6c75c012f13c5

Though I don't know if we still care about that "return to to tiprack" API for demos

mcous, 2018-07-13:
> When a pipette is not assigned to a trash container, it drop_tip should return the tip to the last empty space in the tiprack

I may have missed it, but I don't see this behavior described in any documentation. We also have `return_tip` for returning tip to last location. Dropping tip where it is if it has no location to drop it at seems like correct behavior to me

IanLondon, 2018-07-13:
Yeah it was an undocumented feature in OT1, used in demo protocols. Makes sense to use `return_tip` instead.

closed 2018-07-13 by IanLondon