Opentrons/opentrons #707: Cancel button does not work in app

Laura-Danielle, 2018-01-31:
## overview

Lack of functionality for certain buttons in the app.

## behavior

During a protocol run where the options are 'Pause' or 'Cancel', the 'Cancel' button is non-responsive. 
Seen on windows 10 and Mac OS.

## steps to reproduce

-Calibrate labware
-Run protocol
-Attempt to cancel protocol

Laura-Danielle, 2018-02-05:
@Kadee80 @mcous just to clarify, when I press the 'pause' button and then decide to cancel, the button does nothing. If something goes disastrously wrong in a protocol, would be good if the screen moved back to 'labware set-up' and then homed the robot if someone presses 'Cancel'.

umbhau, 2018-04-26:
Note this should be resolved by 855.

PR #1443 (2018-05-14): fix(app): Send session.resume before stop to cancel paused runs
## overview

The RPC API is unable to handle a `session.stop` call if the protocol run is paused. This PR works around that issue by always calling `session.resume` right before `session.stop`.

Fixes #707

## changelog

- fix(app): Send session.resume before stop to cancel paused runs

## review requests

Needs real robot testing! This will be difficult / impossible to test using virtual smoothie.

- [x] Regular cancel still works (tested on Sunset and Moon Moon)
    1. Start protocol run
    2. Click "Cancel"
    3. Run should c [... cut]
commit e995c9b7 (2018-05-14): fix(app): Send session.resume before stop to cancel paused runs (#1443)

Fixes #707
closed 2018-05-14 by mcous