Opentrons/opentrons #708: Button to re-run a workflow in top right of app shows up after protocol run but does not work

Laura-Danielle, 2018-01-31:
## overview

Lack of functionality for certain buttons in the app.

## behavior

When a protocol is finished running, a button appears to 'Run protocol again'. However, this button does not initiate a new run of the protocol.

If I want to run the protocol again, I have to click back through labware set-up and press 'Run Protocol' in the bottom left hand corner.

## steps to reproduce

-Run through a protocol
-Attempt to press the 'Run Protocol' button in the top right hand corner

Laura-Danielle, 2018-02-05:
Not sure about labeling for this issue @mcous assuming it's an easy fix though.

mcous, 2018-02-08:
Oops forgot about that button. @btmorr there was a time where (I assume due to the robot singleton?) we were assuming that a `session` could only be run once. Is that still the case?

mcous, 2018-04-26:
Closing in favor of #1270

closed 2018-04-26 by mcous