Opentrons/opentrons #709: Connectivity: One Robot at a Time Tooltip

umbhau, 2018-01-31:
As a user, I need to know that I have to disconnect from all robots before connecting to any other robots.

## Acceptance Criteria
- ‘Connect’ button inactive for all robots other than currently connected robot
- User informed that they have to disconnect from all other robots before connecting to a robot (e.g. hover tooltip on connect button)

## Design
- https://projects.invisionapp.com/share/V5FI4I26QWD#/screens/275374451

umbhau, 2018-02-12:
Dependent on a tooltips framework.

mcous, 2018-07-12:
Tooltip framework added in #1855!

umbhau, 2018-07-25:
I am closing this for now. 

Still a valid feature, but not a prioritized one (no reported issues around this yet)

closed 2018-07-25 by umbhau