Opentrons/opentrons #732: PD Basic Pipetting Steps: Consolidate

umbhau, 2018-02-03:
As a protocol designer, I’d like to be able to create a 'consolidate' step, so that I can easily move liquid from many location to one.

## Acceptance Criteria
- User able to select a pipette for consolidate action (from pipettes attached to robot)

### Source Well
- User able to select source well (note well selection modal is not covered in this story)
- User able to set aspiration volume
- User able to select from the following aspirate configurations:
-- 'pre-wet tip'
-- 'touch tip'
-- 'air gap'
-- 'mix' (user able to set # of times and volume) (on every aspirate?)
-- 'disposal volume' (user able to set volume) (???)
-- 'tip change' (user able select 'always', 'once', or 'never')
-- 'well order' (tbd. what are options?)
-- flow rate (tbd. covered in separate user story)

### Destination Wells
- User able to select destination well (note well selection modal is not covered in this story)
- User able to set dispense volume (can this be calculated? e.g., 'set automatically')
- User able to select from the following dispense options:
-- 'mix' (user able to set # of times and volume)
-- 'delay' (user able to set delay in minutes and seconds)
-- 'blow out' (user able to select location [trash, tbd]
-- flow rate (tbd. covered in separate user story)

- User able to expand 'more options' (TBD this is in the mockups but what is it?)

### Style
- Visual style updated to reflect mockup (linked below)
- TBD

umbhau, 2018-02-03:
@howisthisnamenottakenyet 
Do we already have a mockup for the consolidate step? I loosely copied the above requirements from 'Distribute'

IanLondon, 2018-02-07:
# TODO

- [x] All fields created with FE validation (can't save invalid form)
- [x] Visual style updated to reflect mockup (below)
- [ ] User able to open 'more options' modal to see edit step name and notes
- [ ] User able to collapse or expand non-required field actions

PR #778 (2018-02-08): Pd pause & consolidate form + "More Options" modal
## review requests

Check it out: https://s3-us-west-2.amazonaws.com/opentrons-protocol-designer/pd_consolidate-form/index.html

* Create Transfer/Pause/Consolidate form
* Can save options
* Can open "MORE OPTIONS" modal, and edit and save the name/description fields
* Can DELETE STEP in MORE OPTIONS modal

#### Not in scope:

* Consolidate's SubStep items
* Custom names show up in step list (new issue #784)

## overview

Closes #732 

Also, fixes #713 and fixes #726 (this PR includes the now-closed Pause Form PR #750)

## changelog

* Ripped  [... cut]
closed 2018-02-08 by IanLondon