Opentrons/opentrons #742: Factory calibration tool should pre-move to hover above z-height calibration point

btmorr, 2018-02-05:
## overview

Feature request: when CLI tool starts, after homing move to area above deck so that z-height calibration only requires the user to jog in Z

## behavior

**Currently** the user has to jog in X and Y and pick a spot to descend (ambiguous regarding which spot they should pick, and time consuming)

**Desired** the robot homes and then moves to an xyz position above a desired calibration point, and then the user jogs down to the deck

mcous, 2018-06-05:
Fixed in the GUI / HTTP endpoints. Still needs to be addressed in the CLI

btmorr, 2018-09-13:
@andySigler do we want to actually do this? It would be a change for the factory. It might be a good change, but I also know change is difficult so the difficulty might outweigh the benefit.

andySigler, 2018-09-13:
I think as far as change goes this is fairly minor. Previously they were selecting one of the three points, so it would move down, then they'd jog down and press Z to save the height. This change makes logical sense :+1:

sfoster1, 2020-04-27:
This will be fixed in the robot calibration overhaul.

closed 2020-04-27 by sfoster1