Opentrons/opentrons #746: Labware definition file refactor

btmorr, 2018-02-05:
### Problem

Labware definitions are currently stored in a single large json file, and duplicated with some transformations in a sqlite database.

Specific problems:

- There is currently no mechanism for keeping these two databases in sync with each other
- The human-readable datastore (the json file) is not the source of truth at run time
- The labware definitions themselves have some problems in either file, such as the fact that the wells are not given any strict order or organization into or within rows and columns, which has caused unpredictable behavior in user protocols
- The migration process (creating the sqlite database from the json file if it does not previously exist) is slow and unnecessary
- The current state of the server keeps all of both the json datastore and the sqlite database in memory, which nullifies any gains made from keeping the data in a structured database
- It is difficult to incorporate and share user labware definitions

### Proposal

1. Remove the sqlite database from the server
2. Break the single large json file into a directory containing one json file per labware (schemas below)

Laura-Danielle, 2018-02-05:
Not sure if this belongs under this issue, but also making sure that labware definitions are actually correct + naming conventions for each labware include either manufacturing company name and/or model number

IanLondon, 2018-02-05:
Sounds good!!!

Also, we talked about moving labware to the repo root as its own `labware/` project, and adding a barebones package.json so JS projects can import it -- if that's done as a secondary follow-up PR after the initial big refactor labware PR(s), that's OK with me!

@Laura-Danielle maybe model number & manufacturer would go under the `metadata` field?

Something else we talked about was not treating "stackable" modules (like temperature deck) as a normal labware, since it doesn't have any wells. (But it does have an offset that it applied to its child/children, and js world would probably benefit from being able to read that offset...)

@btmorr Did we also talk about being able to create any container that you could make with JSON using `containers.create`? Or is that a separate thing outside this RFC?

btmorr, 2018-02-05:
@IanLondon I actually think @Laura-Danielle 's idea about having separate named definitions per manufacturer is smart, to avoid confusion between one plate and another, but that doesn't change the overall plan at all

I do think that `containers.create` should work with the json schema

btmorr, 2018-02-05:
Also, happy to put the definitions into a repo-level directory with a package.json file. Might as well do it right from the start.

closed 2018-02-06 by btmorr