Opentrons/opentrons #756: No validation when updating container definitions

btmorr, 2018-02-06:
## overview

When updating container offsets, no validation is performed on new values

## behavior

**Current:** When updating container offsets, no validation is performed on new values, which makes it possible to save values that do not make sense, such as points below the deck

**Desired:** When updating container offsets, validate new data

## steps to reproduce

While calibrating labware cause z motor to slip (or otherwise disable it), and click jog until the pipette's z value in pose_tracker is negative, then confirm new setting. The labware definition in the database will be overwritten with a negative z value

mcous, 2019-06-28:
@btmorr is this ticket still applicable?

mcous, 2021-01-26:
Technically, we still don't sanity check labware calibration data. However, this issue has been mitigated by a lot of work in the larger calibration / labware definition system. Closing this ticket in advance of more labware calibration work; sanity checks on labware calibration values may be included in a future ticket if deemed necessary

closed 2021-01-26 by mcous