Opentrons/opentrons #780: Python API:

umbhau, 2018-02-07:
As a robot user, I would like the pipette to hold its z-position between two actions in the same well, so that it can complete protocols more efficiently.

## Acceptance Criteria
- If robot is performing two sequential actions in the same well, the robot should maintain its z-position (i.e., not raise the pipette out of the well) between the actions. 
-- Example: The pipette should remain in the same position when executing the following command, and not move between mixing (one aspirate and dispense) and aspirating its transfer volume:
```
pipette.transfer(
    100,
    plate.wells('A1'),
    plate.wells('A2'),
    mix_before=(1, 50))
```

- As an exception case, the robot *should* move (raise out of the well) if creating an air gap before or after an aspirate.

umbhau, 2018-02-07:
@btmorr 

The above should handle Kristin's request for an 'aspirate with prewet' command.

Worth thinking about eventually creating a 'prewet' alias for 'mix_before' (defaulted to a single aspirate/dispense cycle).

mcous, 2018-09-22:
Issue closed by #839 (please re-open if I'm wrong here)

closed 2018-09-22 by mcous