Opentrons/opentrons #794: Bug: Multichannel does not raise before moving

umbhau, 2018-02-08:
Currently, the multichannel pipette does not raise before moving, potentially crashing into labware.

## Issue: 
Expected: second pipette (multi channel) should raise before moving across the deck
Observed: multichannel does not raise, hitting any labware in the way (esp. tube racks)

## Environment: 
Robot: Mikebot (commit 321538)
Protocol: 
[Serial_dilution_OT2_ekker_v2.txt](https://github.com/Opentrons/opentrons/files/1709162/Serial_dilution_OT2_ekker_v2.txt)

Note: Ben attempting to reproduce with latest branch (including Andy’s fix). May be quick follow-on to Andy’s current work.

mcous, 2018-02-08:
Related (I think): #681

andySigler, 2018-02-09:
This was addressed in PR #774

closed 2018-02-09 by andySigler