Opentrons/opentrons #860: Chang Pipettes: Move Pipette into Position

umbhau, 2018-02-17:
As a user, I would like the pipette carriage to move into an accessible position before I change pipettes, so that I can easily access the pipette screws and plug.

## Acceptance Criteria
- When a user initiates a pipette change (e.g. via # ), the are shown a dialogue box telling them:
-- To remove labware and tipracks 
-- That changing pipettes will clear calibration data
-- To watch out, the robot is about to move
-- (please refer to mockup for actual copy)
- User is able to cancel or continue with pipette change
- If user opts to continue, pipette moves to an accessible location:
-- If changing the right pipette, pipette moves all the way forward, down, and left
-- If changing the left pipette, pipette moves all the way forward, down, and right
-- User shown busy/spinning icon modal while pipette is moving (?)
- Pipette power turned off (to plunger), so that user can safely remove pipette 

## Design
- dialogue box: https://projects.invisionapp.com/d/main#/console/13399396/279798950/preview
- spinning icon: https://projects.invisionapp.com/d/main#/console/13399396/279798949/

umbhau, 2018-03-05:
@mcous @Laura-Danielle 

After discussion with @mcous, we potentially need a new BE API endpoint / function to 'move the pipette to an accessible location'.

umbhau, 2018-03-20:
FYI @mcous 

After discussion with @btmorr, the BE endpoint will expose controls to engage or disengage individual axis, so the Run App will have to be somewhat intelligent and select which axes to disengage before pipette change (in this case, axis b and c)

mcous, 2018-04-24:
Addressed by PRs listed above!

closed 2018-04-24 by mcous