Opentrons/opentrons #861: Change Pipettes: Initiate from Calibration Screen

umbhau, 2018-02-17:
As a user, I would like to be prompted to change my pipette if I currently have the wrong pipette attached.

## Acceptance Criteria
- User sees a visual alert on the calibration page if the currently attached right or left pipette does not match the pipette specified in the current protocol
- If the wrong pipette is currently attached, user is able to click a button to initiate the pipette change.
- Note actual pipette change is covered in #860

## Design
https://projects.invisionapp.com/d/main#/console/13399396/279798952/preview
- Ask to move labware: https://projects.invisionapp.com/d/main#/console/13399396/279798950/comments

## Implementation Notes
- Dependent on knowing which pipette is attached

pantslakz, 2018-02-20:
@umbhau should we also add to story that user is unable to move forward in calibration unless correct pipette is attached?

umbhau, 2018-04-24:
Note decided to implement change pipettes without this feature.

mcous, 2018-04-30:
Closed by #1263

closed 2018-04-30 by mcous