Opentrons/opentrons #870: RFC: Release channel process

mcous, 2018-02-19:
## overview

This is an RFC PR to try to get some direction on our release channel strategy. We don't need everything figured out, but figuring out some broader preferences would help me narrow down current, incremental work.

By release channels, I'm referring to a system where the Run App (and probably other projects as well) release updates on different channels based on stability and release frequency. For example, Chromium releases on [several channels](https://www.chromium.org/getting-involved/dev-channel#TOC-How-do-I-choose-which-channel-to-use-):

> * Stable channel: This channel has gotten the full testing and blessing of the Chrome test team, and is the best bet to avoid crashes and other issues.  It's updated roughly every two-three weeks for minor releases, and every 6 weeks for major releases.
> * Beta channel: If you are interested in seeing what's next, with minimal risk, Beta channel is the place to be.  It's updated every week roughly, with major updates coming every six weeks, more than a month before the Stable channel will get them.
> * Dev channel: Want to see what's happening quickly, then you want the Dev channel.  The Dev channel gets updated once or twice weekly, and it shows what we're working on right now.  There's no lag between major versions, whatever code we've got, you will get.  While this build does get tested, it is still subject to bugs, as we want people to see what's new as soon as possible.
> * Canary build: Canary builds are the bleeding edge. Released daily, this build has not been tested or used, it's released as soon as it's built.

## proposed behavior

At the very least, we will probably want a few channels. For example, a very minimal set could be "stable" (normal release) and "edge" (current development build). Or, we could do "stable, "beta", and "alpha" if we wanted to mirror Chrome's process more closely.

The specific channels we use aren't really the point of this RFC, because whatever the first implementation of this looks like, it'll probably just have one channel. Which leads me to:

## what's the question here?

The question I'm running up against is: how do we determine what channel to release on? In no particular order (and with only some mutual exclusivity):

1. Should the channel be based on the version? e.g., `v1.2.3-alpha.2` -> "alpha" channel
2. Should the channel be based on the branch? e.g. `master` -> "latest", `develop` -> "next"
3. Do we only release if the version number has changed?
    * If so, how technically do we accomplish this?
4. Do we only release if the commit is tagged in git?

mcous, 2018-02-19:
I also have a soft preference for (2). CI checks and process seems like it warrants further discussion and buildout, but at a high level that sounds sane.

I think moving forward, I'm going to do the following release channels to start:

- `v3a` -> "alpha" channel
    - Will not enforce version bumps just yet, but I think that's ok for right now
- Any other branch (e.g. some dev feature branch) -> "${branch_name}" channel
    - Unless someone has a better idea?

IanLondon, 2018-02-19:
If every merge needs to bump the version, will that mean we will update `v3a` less frequently? Or are we ok with a rapidly ticking patch number that increments for each PR?

I want to avoid getting in a place where we are blocking PRs to `v3a` whenever the changes in those PRs wouldn't warrant a new release in themselves, thereby siloing code into a handful of `pre-develop` branches.

Eg, right now Protocol Designer PRs don't affect users since Protocol Designer isn't usable yet. Should PRs for Protocol Designer bump the release? I think we should expect to always have projects in this stage in the monorepo, getting updated frequently but not necessarily making changes for any users (even alpha users).

(PS when are we going to rename `v3a` -> `develop`?)

mcous, 2018-02-19:
I don't think @btmorr is suggesting bumping the version for every PR into the develop branch (which, speaking from personal experience, is unworkable). It seems to me like we would bump in some sort of "release" PR into `master`

mcous, 2018-02-19:
@IanLondon and @btmorr after our talks in person, here's what things are looking like to start. As a reminder, at the moment the Run App is the only project that has user-facing concepts of releases and channels, so all other projects will be unaffected by this process for the time being.

Ping @umbhau so you're aware of this discussion.

- Two user-facing channels: "stable" and "edge"
    - "stable" channel maps to `stable` (or `master`?) branch
    - "edge" channel maps to `edge` (or `develop`?) branch
- To release on "edge" channel:
    - Merge PR into edge branch
    - CI publishes a release (even if version hasn't updated)
- To release on the "stable" channel
    - Merge release PR into `edge` that bumps version number
        - This will release the new version on `edge`
    - Open and merge a PR into `stable` that syncs it with `edge`
    - CI verifies that version files have been updated
    - CI publishes the release
- On any push to another branch, CI will publish a release with the branch name as the channel

We can rename branches as part of this process, or hold off until afterwards; regardless it doesn't affect the scope of work

mcous, 2018-03-27:
Closing this because we implemented a mix of (1) and (4) in #1104

closed 2018-03-27 by mcous