Opentrons/opentrons #895: Distribute command not functioning as expected

Laura-Danielle, 2018-02-22:
## overview

In some instances, the distribute command appears to not be equally dispensing volumes across a plate. Will need to do further testing on:
1. Different types of pipettes
2. Different volumes
3. Liquid types (? not sure if necessary)

## behavior

Uneven volumes between distribute. Sometimes the pipette will not aspirate correct volume from source etc.

## steps to reproduce

Need to do further testing first. Problem was initially seen with the following protocol:
```
from opentrons import containers, instruments

trough = containers.load('trough-12row', '3', 'trough')
plate = containers.load('96-PCR-flat', '2', 'plate')
tubes = containers.load('tube-rack-2ml', '8', 'tubes')

p300rack = containers.load('tiprack-200ul', '1', 'p300rack')
p200rack = containers.load('tiprack-200ul', '4', 'p200rack')

p300_multi = instruments.Pipette(
   name="p300_multi",
   tip_racks=[p300rack],
   min_volume=50,
   max_volume=300,
   mount="left",
   channels=8
)

p200 = instruments.Pipette(
   name="p200",
   tip_racks=[p200rack],
   min_volume=20,
   max_volume=200,
   mount="right"
)

def run_custom_protocol(sample_volume: float=80, buffer_volume: float=40):
   # Transfer buffer to all wells, except row 1 on plate.
   p300_multi.distribute(
       buffer_volume, trough['A1'], plate.columns('2', to='12'))

   # Transfer 8 tube samples to row 1 on plate, and mix.
   p200.transfer(
       sample_volume*2,
       tubes.wells('A1', to='D2'),
       plate.columns('1'),
       new_tip='always',
       mix_after=(3, sample_volume)  # How much volume to mix
   )

   # Dilution transfers with mixing of rows after each transfer.
   p300_multi.transfer(
       sample_volume,
       plate.columns('1', to='11'),
       plate.columns('2', to='12'),
       mix_after=(3, sample_volume)  # How much volume to mix
   )

run_custom_protocol(**{'sample_volume': 80.0, 'buffer_volume': 40.0})
```

umbhau, 2018-03-14:
@Laura-Danielle are you still seeing this issue?

mcous, 2018-09-24:
After talking to @Laura-Danielle, this bug does not appear to be present any longer. Advanced pipette functions still need work generally, but it will be covered in upcoming refactors

closed 2018-09-24 by mcous