Opentrons/opentrons #915: Containers.create not working

umbhau, 2018-02-23:
As a user, I would like to be able to create containers. 

Note: This is blocking the effort to translate OT1 protocols for OT2 (remaining protocols have custom containers)

## Expected Behavior
- User able to create a container by calling containers.create

## Observed Behavior
- Containers.create fails to create container

## Steps to recreate
- TBD. Reported by Jerry and have not yet attempted to recreated.

umbhau, 2018-02-26:
fyi @andySigler @btmorr. 

This is a blocker for updating OT1 protocols to OT2 (we're through 90% of the protocols, but the remaining ones have customer containers).

btmorr, 2018-02-27:
@umbhau I don't know how this used to work, but currently `containers.create` has a parameter named `save` with a default value of `False`. If save is false, the container is returned (so it can be assigned to a variable and used in the protocol) but is not saved to the database. To get the desired behavior, add `save=True` to the parameters.

PR #931 (2018-02-27): Remove save parameter on containers.create
## overview

`containers.create` should always save the definition, because otherwise it is not possible to use the created definition in a protocol, so `save=False` is a non-sense option. Fixes #915 

## changelog

- (bugfix) Remove `save` parameter from `containers.create`, restoring expected functionality
closed 2018-02-27 by btmorr