Opentrons/opentrons #927: P10 Tip Probe Height

umbhau, 2018-02-27:
PLACEHOLDER - Revisit 3/12 with HW when we've received new pipettes.

As a user, I'd like my robot to tip probe from a safe height, reducing the chance of a crash. 

## Acceptance Criteria
- Update tip probe height for P10 such that the pipette clears the probe box if no tip is attached.

## Implementation Notes
- This will also reduce the chance that tip probe is thrown off by the flare at the top of tips
- Will need to test and ensure that raising tip probe height does not result in too much tip deflection
- Note the 3d printed pipettes drop tips more easily, making deflection more of an issue. These will be replaced by polypropylene tips in ~2 weeks. Let's hold on implementing this feature until we're received the new pipettes and confirmed with Taylor.

umbhau, 2018-02-27:
@btmorr @andySigler 

Just want to make you guys aware of this ticket -- potentially need to optimize tip probe height for P10 pipette. Possibly relevant to #817

andySigler, 2018-02-27:
I'm fairly certain this will be fixed by pr https://github.com/Opentrons/opentrons/pull/838

umbhau, 2018-02-27:
@andySigler excellent. Will re-test after #838 is merged and close this one if that's the case.

closed 2018-03-27 by Laura-Danielle