Opentrons/opentrons #934: Multichannel Tip Pickup

umbhau, 2018-02-28:
As a user, I would like to be able to repeatedly pick up tips with a multichannel pipette.

## Expected Behavior
- Multichannel pipette should pick up one row after another of tips

## Observed Behavior
- Multichannel pipette correctly picks up one row of tips, then moves to the wrong location to pick up the second row of tips

## Steps to Recreate
- Connect to any (?) robot (tested with Muddy-Resonance and others)
- Load following protocol
-- Note protocol includes two options for picking up tips, "for tip in rack" and "for tip in racks.cols():". Bug observed with both methods. Be sure to comment one out before running.
- Attempt to run protocol
- Observe robot:
-- (1) correct picks up first row of tips
-- (2) moves in the wrong direction to pick up the second row of tips (the back of the pipette moves to ~the H1 tip in the rack, instead of the A2 tip, as expected).

```
"""
Author: Carlos Fernandez

#Python 3.5
#Imported Libraries

"""
from opentrons import labware, instruments,robot
rack_slot = '1'
rack = labware.load('tiprack-200ul',rack_slot)
#P10 Single
pipette = instruments.P300_Multi(mount='left')
pipette.set_flow_rate(aspirate=200, dispense=300)
#pipette = instruments.Pipette(name ='p301-multi', mount = 'left', channels = 8)
#pipette_2 = instruments.Pipette(name = 'p301', mount = 'right')
pipette.pick_up_current = 0.2
#pipette_2.pick_up_tip(rack.well('A1'))
#pipette_2.return_tip()
volume = 10

## Option 1:

for tip in rack:
    print("tip: ", tip)
    pipette.pick_up_tip(tip, presses = 1)#Pick up tip
    pipette.drop_tip()
    print(pipette.drop_tip(rack.top(-30)))

## Option 2

for tip in racks.cols():
    print("tip: ", tip)
    pipette.pick_up_tip(tip, presses = 1)#Pick up tip
    pipette.drop_tip()
    #print(pipette.drop_tip(rack.top(-30)))
```

umbhau, 2018-02-28:
@btmorr , @andySigler 

This bug needs a bit more testing, but wanted to give you a heads up since @Carlos-fernandez reported it and it is blocking his testing. Will continue to try and reproduce tomorrow.

Laura-Danielle, 2018-03-02:
It was something wrong with the script not the API. Closing this issue.

closed 2018-03-02 by Laura-Danielle