Opentrons/opentrons #962: PD Liquid Tracking: Highlight Involved Wells

umbhau, 2018-03-03:
As a protocol designer, I'd like to see the which wells are involved in an action

## Acceptance Criteria
- When user hovers over a sub-step (e.g. one individual transfer within a consolidate), source and destination wells are highlighted on deck map

## Design
- @howisthisnamenottakenyet: 
-- Assuming this should be on hover, not on click
---- The above assumption changes if you need to click on a step to see the initial deck state (i.e., if a user could click on one card to see initial deck state for that step, but then hover over a transfer within another step to simultaneously see the wells involved in the second step)
-- Assuming mockup shows highlight style (color, border, etc.)

IanLondon, 2018-03-23:
I labeled this as `blocked` b/c it's too big to be actionable

#1083 addresses part of this

umbhau, 2018-03-26:
@IanLondon updated this story to include 'step' only, and not 'sub-step'

IanLondon, 2018-04-05:
After paring this ticket down, it became a duplicate of #1083

closed 2018-04-05 by IanLondon