Opentrons/opentrons #990: Robot Wifi Not Appearing on Windows

umbhau, 2018-03-07:
PLACEHOLDER - getting details.

## Expected: Wifi networks should appear in networks dropdown

## Observed: Wifi networks do not appear in windows machines 

## Steps To Recreate
Note: Have not recreated this one yet; reported by @Laura-Danielle 
1. Open run app on Windows (version >= 2018-03-06)
2. Open robot settings tab
3. Not wifi networks not visible in dropdown?

umbhau, 2018-03-07:
@mcous @Kadee80 

Just an FYI -- not sure if this is related to any ongoing connectivity work.

mcous, 2018-03-07:
@umbhau ongoing connectivity work as not touched existing discovery logic. @Laura-Danielle what version of Windows and which exact build did you see this behavior on?

I'm unable to reproduce this on my Windows 10 (v10.0.16299) with latest `edge` (commit 148d316 at time of writing)

umbhau, 2018-04-05:
@mcous Reproduced this again on @Carlos-fernandez laptop:
- Last stable build: https://s3.amazonaws.com/opentrons-app/builds/Opentrons-v3.0.0-beta.2-win-b3193.exe
- PP2 Robot: OPENTRONS-OT2P20180327A04
- Was able to successfully connect on my mac

commit b4737c48 (2018-08-22): feat(app): Enable support for IPv4 wired robots by default

Fixes #990, fixes #1964
PR #2090 (2018-08-22): feat(app): Enable support for IPv4 wired robots by default
## overview

This PR enables "new" discovery to support the container changes in #1970 by default in the app. It also adds the changes made in #2072 to the top-level Makefile so that `make push-api` and `make term` from the repo-level will automatically work with whatever ethernet robot you have plugged in (whether its a new IPv4 or an old IPv6 one).

Presumably (based on extensive in-house testing and positive results with users who were experiencing issues) fixes #990 and fixes #1964. This is the last known app-side blocker for the ` [... cut]
commit d3a3afac (2018-08-22): feat(app): Enable support for IPv4 wired robots by default (#2090)

Fixes #990, fixes #1964
closed 2018-08-22 by mcous