Opentrons/opentrons #999: Change Pipettes: home pipette plunger and z-gantry HTTP endpoint

mcous, 2018-03-08:
## overview

As a robot user, I would like to use the Run App to home a newly installed pipette so that I can verify I have installed the pipette correctly.

For the Run App to tell the robot to home a given pipette, the HTTP API needs an endpoint that the app can use to home the pipette on a given mount

## acceptance criteria

- API has an endpoint that can call `pipette.home` on a given mount

### dependencies

- Get pipette info (PR #998)

### dependents

- Test new pipette (PR #865)

## specifications

Actual spec TBD by FE + BE requirements review

- POST endpoint
    - If command time (e.g. `home`) is in the payload, can be shared with endpoint described in #866 
- Returns 200 if homing is successful
- Returns 4xx with descriptive error message if not allowed (e.g. no pipette in mount)
- Returns 500 with descriptive error message if homing fails

## test plan

- No pipette installed in left
    - [ ] POST home for left responds 4xx
- No pipette installed in right
    - [ ] POST home for right response 4xx
- Pipette installed in left
    - [ ] POST home for left responds 200
    - [ ] Left pipette homes its plunger
    - [ ] Left pipette homes its z-gantry
    - [ ] GET pipette info shows left pipette enabled
- Pipette installed in right
    - [ ] POST home for right responds 200
    - [ ] Right pipette homes its plunger
    - [ ] Right pipette homes its z-gantry
    - [ ] GET pipette info shows right pipette enabled
- Induce home failure somehow (e.g. yank power to smoothie)
    - [ ] POST home for right responds 500
    - [ ] Subsequent POST home (with induced failure condition removed) succeeds

btmorr, 2018-04-04:
The error states are not precisely as described in this ticket, but there is a `home` endpoint for pipettes added in #1100. Could either @umbhau or @mcous take a look and either close or modify this ticket? Also, do we actually want it to reject a home request if the pipette is not detected? This would prevent us from using this endpoint to home an uncommissioned pipette, or to home a mount without a pipette actually attached.

btmorr, 2018-04-04:
Oh, I see. #1100 referenced this ticket, but didn't use the "fixes" keyword to automatically close the issue. Would still appreciate the double-check on requirements

mcous, 2018-04-10:
Closed by #1100

closed 2018-04-10 by mcous