Opentrons/opentrons #1011: Define 8-channel access type in labware definitions

IanLondon, 2018-03-12:
The new JSON format in `labware-definitions` has made  well order more explicit: under the `ordering` key is a 2D array that gives wells by column `[column][row] -> well`.

However, we can still only infer how 8-channel pipettes can or cannot access each labware type.

For PD, we need to be able to determine, for each labware type:
1. If the labware is accessible or inaccessible to an 8-channel pipette (eg user should not be able to use 8-channel with a flask, tube rack, etc that will just cause a crash)
2. For liquid tracking: which tips go into which wells (Ie, when "Tip 1" is in a given well, what wells are the other tips in?)

I think there are 4 "8-channel access types":

1. **96 format** Tip 1 of the 8-channel goes into the `A` row only, then Tip 2 is in `B`, Tip 3 in `C`, and so on.
2. **384 format** Staggered access: when Tip 1 of the 8-channel goes into the `A` row, Tip 2 is in `C`, Tip 3 is in `E`, etc. When Tip 1 of the 8-channel goes into the `B` row, Tip 2 is in `D`, 3 in `F`, etc.
3. **"Trough"** format. All tips fit in the same well at once. There should only be a single `A` row. In some labware, there will only be a single A1 well, in others like `"trough-12row"` there will be A1-A12. *(Aside: aspirating X uL removes X uL from the well, which delivers X/8 uL into each of the 8 tips. Dispensing X uL adds X uL to the well, removing X/8 uL from each tip.)*
4. **"Not accessible"** - tube racks, flasks, 6/12/24/48 well formats, `"point"`, etc. The 8-channel cannot use these labware types.

# Proposal

So I propose we add a new key to the labware JSON files, `"8-channel-access-format"` that takes the value `"96"`, `"384"`, `"trough"`, or `false`.

# Alternative

Instead, we could use the array under the `ordering` key to infer the type, eg single row is probably "trough", 12x8 is probably 96-format, 24x16 is probably 384-format, anything else is probably not accessible to 8-channel. The downside I see with this is that some labware will be inferred as the wrong format: a flask will appear to be a trough, a staggered 12x8 will look like a 96-grid, a single-row tube rack will look like a trough.

-----

Thoughts?

mcous, 2018-03-12:
Gonna turn this over in my head for a little and see if anything smart falls out. Just what's immediately off the top of my head:

* How is @Opentrons/py planning on handling 8-channel access on the server side?
* Could we infer based on geometry? i.e. actual well locations rather than well ordering
    * Technically we have all the information necessary
    * Might be much harder to implement than an explicit top-level key though
    * Easier to do an `access` key and then deprecate / stop using it when we figure out geometry than vice versa
* Are we ever planning on supporting 6, 12, or 16 channel pipettes? 
* For the value of the access type, perhaps we could use the stagger distance to be a little more generic?
    * e.g. `trough` -> `0`, 96` -> `1`, `384` -> `2`

IanLondon, 2018-03-12:
@mcous if we have the definitions right, geometry will work nicely. I have a gist in 50 lines of code https://gist.github.com/IanLondon/ce55b037294ba4ef255f109eaee16d60

Seems to work with the containers I looked at, example output for Tip 1 in A1, then A2, then B1 (it returns `null` if any tip doesn't match a well):

```
96-flat A1 [ 'A1', 'B1', 'C1', 'D1', 'E1', 'F1', 'G1', 'H1' ]
96-flat A2 [ 'A2', 'B2', 'C2', 'D2', 'E2', 'F2', 'G2', 'H2' ]
96-flat B1 null
384-plate A1 [ 'A1', 'C1', 'E1', 'G1', 'I1', 'K1', 'M1', 'O1' ]
384-plate A2 [ 'A2', 'C2', 'E2', 'G2', 'I2', 'K2', 'M2', 'O2' ]
384-plate B1 [ 'B1', 'D1', 'F1', 'H1', 'J1', 'L1', 'N1', 'P1' ]
trough-12row A1 [ 'A1', 'A1', 'A1', 'A1', 'A1', 'A1', 'A1', 'A1' ]
trough-12row A2 [ 'A2', 'A2', 'A2', 'A2', 'A2', 'A2', 'A2', 'A2' ]
trough-12row B1 null
tube-rack-.75ml A1 null
tube-rack-.75ml A2 null
tube-rack-.75ml B1 null
trash-box A1 [ 'A1', 'A1', 'A1', 'A1', 'A1', 'A1', 'A1', 'A1' ]
trash-box A2 null
trash-box B1 null
T25-flask A1 null
T25-flask A2 null
T25-flask B1 null
fixed-trash A1 [ 'A1', 'A1', 'A1', 'A1', 'A1', 'A1', 'A1', 'A1' ]
fixed-trash A2 null
fixed-trash B1 null
```

IanLondon, 2018-03-13:
Closed by #1019

closed 2018-03-13 by IanLondon