Opentrons/opentrons #1017: Change Pipettes: Move to Location (API)

umbhau, 2018-03-13:
As a user, I would like to be able to move my pipette to an easily accessible location

## Acceptance Criteria
- [ ] New endpoint to expose robot-specific location info
- [ ] Endpoint includes move-to coordinates

btmorr, 2018-03-16:
Difficulty with this endpoint: for the move to change pipette, the pipette must be low enough that the user can access the screw holes, which is a specific height for the mount, but we _usually_ issue move_to commands on a pipette. We're not bound to that--we could do a driver-level move command based on the mount, but this would probably not have checking built in to *ensure* that the pipette does not collide with the deck. This shouldn't be a problem, because we shouldn't be building pipettes that are so long that the robot can't go to pipette change position without colliding with the deck, but something I wanted to raise.

Also, if we issue the move based on mount position calculations, this make it conceptually easier in some ways (you don't have to check which pipette is installed, for instance), but it makes some things harder, because we don't currently have high-level move_to commands for this. No problem with building them as part of this, but it's a thing to do. It also requires the caller to think about whether the `move` command they are requesting is a mount move or a pipette move, which I think is the biggest risk. Would appreciate feedback @umbhau @mcous @andySigler @TaylorGoodman

btmorr, 2018-03-16:
Proposal: provide an `info` endpoint that returns data like this:

```
{
    'positions': {
        'change_pipette': {
            'target': 'mount',
            'left': (100, 100, 50),
            'right': (120, 100, 50)
        }
    }
}
```

The `target` field would tell whether it's an action on a "mount", "pipette", or something else, and then the rest of the data would provide whatever info makes sense.

umbhau, 2018-03-16:
Ben and I spoke offline -- we both think it's worth the effort to implement a new mount-specific move-to endpoint. For the record, this would leave us with:
- One pipette move-to that allows you to specify move-to points for the pipette with or without tip
- One mount move-to that allows you to specify move-to points for the mount without a pipette
...which does imply some intelligence on the FE.

PR #1059 (2018-03-22): Add position info and move endpoints
## overview

Add HTTP endpoints needed for move, with info for moving a pipette to attach tip position and moving a mount to correct position for swapping pipettes (correct positions determined experimentally). Fixes #1017 and fixes #1009 

## changelog

- (feature) Add endpoint with info for correct position data
- (feature) Add move endpoint for both pipette and mount targets
- (bugfix) Add port back into api `make push` command
- (feature) Add api `make restart` command

## review requests

Mount move is still questionable. Both modes need to be tested o [... cut]
closed 2018-03-22 by btmorr