Opentrons/opentrons #1041: 8-Channel Crash into Tiprack

umbhau, 2018-03-15:
8-channel pipette observed crashing into tiprack when traveling between two well plates located above and to the side of the tiprack.

## Expected Behavior
- Pipette should move to max height (defined in this case by standard tiprack height) before moving in xy

## Observed
- Pipette crashes into corner of tiprack when traveling between two wells
- Video: https://www.youtube.com/watch?v=oILGefK2yxg&feature=youtu.be
- [QA Bug Report](https://docs.google.com/document/d/1cqy9V2t7VwaP3bBbVU7o6MQGaxAU1lVhRPlhL5zjbwM/edit#

## Steps to Recreate
- Open app version 3.0.0 (3.0.0.6807)
- Connect to robot 'cool-flower' on commit 7c386d5
- Load protocol 'Plate Mapping HC'
- Properly load 
-- P200 tip rack in slot 4
-- 96 PCR Flat in slot 5 and 7

Reported by Karen on 2018-03-14)

## Protocol

```from opentrons import containers, instruments

p200rack = containers.load('tiprack-200ul', '4')
trash = containers.load('trash-box', '10')

source_plate = containers.load('96-PCR-flat', '11')

dest_plates = [
    containers.load('96-PCR-flat', '3'),
    containers.load('96-PCR-flat', '6'),
    containers.load('96-PCR-flat', '9'),
    containers.load('96-PCR-flat', '8'),
    containers.load('96-PCR-flat', '5'),
    containers.load('96-PCR-flat', '7')
]

p200_multi = instruments.P300_Multi(
    mount="left",
    tip_racks=[p200rack]
)

p200_multi.pick_up_tip()

def run_custom_protocol(odd_volume: float=45, even_volume: float=90,
                        number_of_destination_plates: int=6):
    if number_of_destination_plates > len(dest_plates):
        raise RuntimeError((
            'Number of destination plates is too high. {} was specified, ' +
            'but the max is {}').format(
                number_of_destination_plates, len(dest_plates)))

    # map odd_volume to all odd rows of all 6 destination plates
    for i in range(0, 12, 2):
        target_rows = [plate.cols(i) for plate in dest_plates]
        p200_multi.distribute(odd_volume, source_plate.cols(i), target_rows)

    # map even_volume to all even rows of all 6 destination plates
    for i in range(1, 12, 2):
        target_rows = [d.cols(i) for d in dest_plates]
        p200_multi.distribute(even_volume, source_plate.cols(i), target_rows)

run_custom_protocol(**{'odd_volume': 45.0, 'even_volume': 90.0, 'number_of_destination_plates': 6})
```

umbhau, 2018-03-15:
@btmorr @Laura-Danielle 
Bug I mentioned last night, in which an 8-channel crashes into a tip rack when traveling between well plates. Tiprack involved in crash is the tallest labware on the deck, and it is used in the protocol. Bug is in the backlog for now.

Laura-Danielle, 2018-03-28:
Couldn't replicate.

closed 2018-03-28 by Laura-Danielle