Opentrons/opentrons #1092: PD Error Handling: Aspirate Outside Pipette Capacity

umbhau, 2018-03-26:
As a user, I should not be allowed to aspirate a volume of liquid outside a pipette’s recommended range

## Acceptance Criteria

- Volume field should throw an error if user attempts to enter a value lower or higher than the pipette’s maximum volume
- Validation shown on deselect
- User prevented from saving
- 

## Design
- TBD @howisthisnamenottakenyet 
- lStyle for field-level errors

IanLondon, 2018-05-07:
For **consolidate & distribute**, if the total aspirate volume* is greater than the pipette's max recommended volume, then the consolidate/distribute acts identical to a transfer (transfer is aspirate-dispense, aspirate-dispense instead of asp-asp-disp / asp-disp-disp). To me, that doesn't make sense to do, it's confusing that a consolidate / distribute has devolved into a transfer.

For example, "with P10, consolidate A1 + A2 -> B1, 25uL` looks like

```
aspirate 10uL A1 (max)
dispense 10uL B1
aspirate 10uL A1
dispense 10uL B1
aspirate 5uL A1
dispense 5uL B1

...on to A2
```

That said, if **transfer** is given a total aspirate volume greater than pipette max, it can just split it up:
A 25uL transfer from A1->B1, A2->B2 could be:

```
A1 -> B1 10uL, A1 -> B1 10uL, A1 -> B1 5uL
A2 -> B2 10uL, A2 -> B2 10uL, A2 -> B2 5uL
```

*Total aspirate volume = given volume + disposal volume (if any)

umbhau, 2018-06-11:
Closing in favor of #1593.

closed 2018-06-11 by umbhau