Opentrons/opentrons #1099: Robot Default Configs: Tip Pickup

umbhau, 2018-03-26:
PLACEHOLDER

As a user, I would like my robot to use a tested, effective method for picking up tips.

## Acceptance Criteria
- Robot uses default method for picking up tips, as specified in robot default configs doc (below)
- Default includes specified method (# of presses, and increment in distance per press) and current
- Defaults are set per pipette model

## Implementation Notes
- How might we ensure defaults persist across robot updates?
- What units do we want to standardize on?

## Resources
- [Robot config spreadsheet](https://docs.google.com/spreadsheets/d/1KOGUng337g-blA8Kw40QwSXYjWRodoclLBR8LQOQYd0/edit#gid=749988930) (PLACEHOLDER - pending final data)

umbhau, 2018-03-30:
@andySigler drop tip story we talked about. 

Just pending a results from the drop-tip test to confirm an 0.6 A pick-up current doesn't cause any downstream problems.

btmorr, 2019-01-04:
This is either done or irrelevant

closed 2019-01-04 by btmorr