Opentrons/opentrons #1128: Calibration: Calibrate Labware with Assigned Pipettes

umbhau, 2018-03-30:
PLACEHOLDER/RFC 

As a user, I would like to calibrate my labware with the pipette that will use it, to ensure I get an accurate calibration.

It's hard to calibrate a tuberack with a multi-channel.

## Acceptance Criteria
- If a labware is only used with one pipette, calibrate it with that pipette.
- If a labware is used by both pipettes, calibrate it with the single-channel pipette.
-- If that is ambiguous, calibrate it with the left pipette

## Implementation Notes / Open Questions
- Is it worth calibrating with both pipettes, if they are different lengths (e.g. what if user used non-standard tips)?
- Should we simplify, and just say 'calibrate tube rack with single channel'?

pantslakz, 2018-04-26:
@umbhau is this the same as https://github.com/Opentrons/opentrons/issues/1264 ?

umbhau, 2018-04-26:
@pantslakz Yes, thanks for the catch! Closed #1264 and migrated details over here.

umbhau, 2018-05-09:
@mcous good questions on a very ambiguous user story. see updates above.

PR #1438 (2018-05-11): feat(app): Order tipracks calibrated by 8-channel first in list
## overview

This PR changes the order of the internal labware list to sort tip racks according to their associated calibrator pipette. Tipracks calibrated by the multi-channel are placed first in the list. This ensures that by the time the user gets to non-tiprack calibration, they will probably be using a single-channel pipette.

Fixes #1128

## changelog

- feat(app): Order tipracks calibrated by 8-channel first in list

## review requests

- Load a protocol with a single and a multi (like `calibration-validation.py`)
- [ ] Tip [... cut]
commit 87daf3f6 (2018-05-11): feat(app): Order tipracks calibrated by 8-channel first in list

Fixes #1128
closed 2018-05-11 by btmorr