Opentrons/opentrons #1144: Labware: Update Labware Definitions

umbhau, 2018-04-03:
As a user, I would like my robot to use the new labware definitions that Henry painstakingly measured.

## Acceptance Criteria
- Update the base labware definitions to the dimensions provided by Henry.

btmorr, 2018-04-16:
Rerun the migration script that was used to create the new base labware definitions and compare the results to the old values. Then use the new definitions on a robot and compare the behavior. For the labware definitions that were updated, default calibrations should be correct (or closer to correct)

emernic, 2018-04-17:
Are the old designs for tube racks shown here: [https://github.com/Opentrons/otone_hardware/tree/master/models](https://github.com/Opentrons/otone_hardware/tree/master/models) still accurate? Or are there new designs coming out soon? I am about to cut some acrylic for various tube racks.

umbhau, 2018-05-22:
@btmorr I believe this is done -- could you confirm?

btmorr, 2018-09-13:
Closing in favor of prioritizing new labware system

closed 2018-09-13 by btmorr