Opentrons/opentrons #1153: Robot Update: Display Dialogue Box on Page View

umbhau, 2018-04-05:
As a user, I would like to be notified that a robot update is available as soon as I view the Robot Settings page.

## Acceptance Criteria
- If a robot update is available, display the 'Update Available' dialogue box as soon as the user views the Robot Settings page.

## Design
- Can use the same dialogue box that we show when a user clicks the 'Update' button:
https://projects.invisionapp.com/d/main#/console/13154070/275374445/preview

umbhau, 2018-04-05:
@mcous missing robot update user story (update notification dialogue box) I mentioned earlier this week.

Kadee80, 2018-06-04:
Would like to revisit this ticket with Product and UX again. 

It makes perfect sense when its 1 person, 1 robot to immediately prompt an update when an update is available upon initially viewing that robots settings (not connected).

My initial implementation rendered app update modal only if the user connected to the robot. This is not what the ticket states. However, while testing in the office with multiple robots in use by several users over wifi I connected to a robot I knew needed an update due to the notification icon in the robot list. I connected to that robot to view its status, and it was in the middle of a run. 

Currently, clicking the [update] button in the  robots info card on the settings page allows a user to update a robot they have not connected to check the status of. For example, someone else is calibrating or in the middle of a run. 

Curious on thoughts about implementing before we make update robot a guarded action.

I'm bumping this up to a `blocked` `large`  as there is some necessary plumbing work at the very least to make sure we store ignore this update correctly and without the current hacky workaround available at this time.

PR #1745 (2018-06-25): feat(app): Show robot available update modal on page view
## overview
This PR closes #1153. If latest ignored update from server does not equal current update, show update available modal on robot page view. Clicking not now will set that available update to ignored in server and the modal will not render again for that robot until a newer update is available.

[image]

## changelog

- add ignoreUpdate requests to API state, store response in server.ignored
- get latest set ignored update via `GET server/update/ignore`
- set current available update ignored `POST server/update/ignore`
- show u [... cut]
commit 185cb1d3 (2018-06-25): feat(app): Show available update modal on  robot page view (#1745)

closes #1153
closed 2018-06-25 by Kadee80