Opentrons/opentrons #1196: PD Deck Setup: Well Selection - Well labels restyling

IanLondon, 2018-04-13:
## overview

[image]

## behavior

Update `Plate` and `SelectablePlate` so that:

* Well labels can be outside of the plate
* Well labels highlight bright blue when wells are selected

## TBD

* 384 reconsideration (number labels bleed together, should we set apart every other label by tint or something?)
* Any mouseover behavior?
* Only 1 well, or all?

howisthisnamenottakenyet, 2018-05-21:
@IanLondon 
(1) Putting well labels outside of the labware should resolve issues with odd labware like the wierdo tuberack. 
(2) I think making the labels blue should only be a thing for hover/click and drag. 

Is there anything missing from designs?

IanLondon, 2018-05-21:
@howisthisnamenottakenyet Cool, I think it's all good except: will this work for 384 plate?

howisthisnamenottakenyet, 2018-05-21:
@IanLondon Wouldn't it look pretty much the same except the text would be white and slightly grey instead of black and slightly grey?

IanLondon, 2018-05-21:
@howisthisnamenottakenyet so I should do even white & odd light gray for 384?

closed 2022-09-30 by bndo