Opentrons/opentrons #1228: Calibration Not Persisting To Terminal

umbhau, 2018-04-19:
Labware calibrations performed in Run App are not available when running via terminal

## Expected Behavior
- User can calibrate labware via run app, and then use those calibrations when running robot directly from terminal

## Observed Behavior
- User calibrates labware via run app, but calibrations are not used when running from term

## Steps to Recreate (as per Kinnari)
- Open run app (Opentrons Version 3.0.0-beta.7 (3.0.0-beta.7.7618)
- Connect to robot (tested on Bitter-Wood)
- Open a protocol and perform labware calibration
- Close run app
- Open terminal
- Attempt to aspirate from calibrated labware and observe robot ignoring calibrations.

umbhau, 2018-05-22:
@btmorr - can close this one out, right?

btmorr, 2018-06-01:
This appears to have been introduced by tip probe, which has been made optional by #1558

closed 2018-06-01 by btmorr