Opentrons/opentrons #1230: PD Basic Pipetting Steps: Well Ordering

umbhau, 2018-04-19:
As a user, I would like to be able to choose between row-first and column-first transfers

## Acceptance Criteria
- User can toggle selection as part of step editor for transfer, distribute, and consolidate steps
- User can choose columns-first or rows-first approach to transfers
- User can choose left-to-right (A1, A2, A3...) or right-to-left (A12, A11, A10)
- User can choose top-to-bottom (A1, B2, C1...) or bottom-to-top (H1, G1, F1)

## Design
- @howisthisnamenottakenyet

umbhau, 2018-08-13:
@howisthisnamenottakenyet Just need to add 'cancel' to the modal.

umbhau, 2018-08-13:
@howisthisnamenottakenyet Let's add copy to make it clear that you FIRST do the first field and THEN do the second field.

e.g.
'top to bottom' THEN 'left to right'

howisthisnamenottakenyet, 2018-08-13:
@umbhau already done

commit 0f0ce163 (2018-08-27): feat(protocol-designer): well ordering advanced step setting

Allow PD users to specify the travsersal strategy between multiple selected wells within a step.

Closes #1230
PR #2143 (2018-08-28): feat(protocol-designer): well ordering advanced step setting

## overview

Allow PD users to specify the travsersal strategy between multiple selected wells within a step.

Closes #1230

## changelog

- Add well order advanced setting to Transfer (asp & disp), Distribute (just disp), Consolidate (just asp), and Mix

## review requests

Please make a protocol where you select multiple wells in one of the compatible steps.  Then change the well order field and confirm
commit d67a7d32 (2018-08-28): feat(protocol-designer): well ordering advanced step setting (#2143)

Allow PD users to specify the travsersal strategy between multiple selected wells within a step.

Closes #1230
closed 2018-08-28 by b-cooper