Opentrons/opentrons #1241: Pipette not dropping tips

umbhau, 2018-04-20:
P300S stopped dropping tips

## Expected Behavior
Pipette plunger should travel low enough to drop tips

## Observed Behavior
Pipette plunger not traveling low enough to drop tips

## Steps to Recreate
- Observed on two beta bots (bindong and kaja) and recreated on Sunset
- Open App version Version 3.0.0-beta.8 (3.0.0-beta.8.7695)
- Connect to robot 'Sunset'
- Run below protocol 
- Observe P300S fail to drop tips

```
from opentrons import labware, instruments, robot

"""
QC Test 
This test is meant to check basic robot functionality
"""

# Set up labware
tiprack = labware.load('tiprack-200ul', '1')
tiprack2 = labware.load('tiprack-10ul', '4')
trough = labware.load('trough-12row', '2')
plate = labware.load('96-flat', '3')
plate2 = labware.load('96-flat', '10')

# Set up pipettes
pipette_multi = instruments.P10_Multi(
    mount='left',
    tip_racks=[tiprack2]
)

pipette_single = instruments.P300_Single(
    mount='right',
    tip_racks=[tiprack]
)

pipette_multi.plunger_positions['drop_tip'] = -6

pipette_single.pick_up_tip()

# Distribute 100ul to starter wells
pipette_single.distribute(
    100,
    trough.wells('A1'),
    plate.wells('A1', 'C1', 'E1', 'G1'),
    trash=False,
    new_tip='never'
    )

robot.pause()

# Transfer 50ml with a 40ul mix after
pipette_single.transfer(
    100,
    plate.wells('A1'),
    plate.wells('B1'),
    mix_after=(3, 40),
    trash=False,
    new_tip='never'
    )

robot.pause()

# Transfer 50ml with a touch tip after 
pipette_single.transfer(
    100,
    plate.wells('C1'),
    plate.wells('D1'),
    touch_tip=True,
    trash=False,
    new_tip='never'
    )

robot.pause()

# Transfer 50ml with a blow out after
pipette_single.transfer(
    100,
    plate.wells('E1'),
    plate.wells('F1'),
    blow_out=True,
    trash=False,
    new_tip='never'
    )

robot.pause()

# Transfer 50ml with an air gap
pipette_single.transfer(
    100,
    plate.wells('G1'),
    plate.wells('H1'),
    air_gap=True,
    trash=False,
    new_tip='never'
    )

robot.pause()

# Consolidate liquid back to trash and continue
pipette_single.consolidate(
    100,
    plate.wells('B1', 'D1', 'F1', 'H1'),
    trough.wells('A1'),
    trash=False,
    new_tip='never'
    )

pipette_single.return_tip()

pipette_multi.pick_up_tip()

# Distribute 50ul 
pipette_multi.distribute(
    5,
    trough.wells('A1'),
    plate2.cols('1', length=12),
    trash=False,
    new_tip='never'
    )

robot.pause()

pipette_multi.consolidate(
    5,
    plate2.cols('1', length=12),
    trough.wells('A1'),
    trash=False,
    new_tip='never'
    )

pipette_multi.return_tip()

robot.home()
```

btmorr, 2018-05-02:
This is the result of a hardware change on the P300 Single and an associated change to the drop tip position--primary fix is to replace pipettes with the old hardware. Workaround:
```
p10.plunger_positions['drop_tip'] = -6
```

umbhau, 2018-05-03:
Tested and confirmed root cause posted by @btmorr above. Closing this issue.

closed 2018-05-03 by umbhau