Opentrons/opentrons #1260: API: UI now expects `pipette.name` to be a valid model

mcous, 2018-04-25:
## overview

To build #861 (View attached pipettes in tip probe), the app needs to be able to compare an RPC pipette (which is effectively just a mirror of an API `Pipette` object) with the response from `GET /pipettes`.

* `GET /pipettes` returns a `model` string
* RPC `Pipette` has a `name` string

When using the "standard" pipette constructors (e.g. `P300_Multi`), the name field of the pipette is set to the `model` string. In the absence of any other way of getting the model from an API pipette, the app is now relying on this relationship to hold.

## possible solutions

- Deprecate explicit `name` parameter in `Pipette` constructor
- Deprecate custom `Pipette` constructor in protocols
- Add `model` field to API `Pipette` object and use that instead of `name` in the app
- Add validation to `Pipette` constructor to force `name` to be a valid `model`

Laura-Danielle, 2018-04-26:
Adding my 2 cents, I think we should deprecate the name parameter within the pipette. It's kind of pointless since now all pipettes are opentrons models.

mcous, 2018-06-05:
Closing because the `Pipette` constructor has been deprecated in the API (in favor of the model factory methods, that set `name` to a correct model) and docs have been updated to reflect this

closed 2018-06-05 by mcous