Opentrons/opentrons #1285: Resize top level nav and title bar

howisthisnamenottakenyet, 2018-04-27:
As a user I would like Protocol Designer to be easier to read

--

## Background
A good chunk of our users actually have tiny screens. [See spreadsheet from lucy](https://docs.google.com/spreadsheets/u/5/d/13nQskpWOwnDVaTRrkW9y7iwsxUhyL3Eoj9SeJN00JwM/edit?usp=sharing)
We can increase room for the sidebar and page by reducing the width of the top level nav and the height of the title bar. 

## Acceptance Criteria
-In wires top level nav width was reduced from 72 -> 60
-In wires title bar was reduced from 56 -> 48
-At this time additional width was given to the side panel but not to the page.

## Design
[image]
[image]
[image]

pantslakz, 2018-04-29:
@IanLondon @Kadee80 note this change should reflect in both run app and protocol designer

mcous, 2018-04-30:
@pantslakz @howisthisnamenottakenyet can you ping this thread when these changes have been uploaded to the components library in Zeplin?

@pantslakz do you think these changes should bu grouped with the text alignment stuff with the titlebar we discussed last week?

pantslakz, 2018-05-01:
@mcous yes this is related to that - but should be sorted now that you have the new component library. as a general rule:

Title | Subtitle is always left aligned, unless there is a "back" or "exit" button, in which case it is then right aligned. 

Zeplin now contains a component for both versions.
Please let me know if this requires any additional clarification, or if we should make a separate ticket to tackle it.

commit 94c249b9 (2018-06-25): fix(components): Resize main navbar and title bar to match designs

closes #1285
PR #1757 (2018-06-25): fix(components): Resize main navbar and title bar to match designs
## overview

closes #1285 by shrinking the width of the main `NavBar` and default height of `TitleBar`. Some additional css was needed to get the layout closer to designs. 

[image]
[image]

## changelog
- shrink `navbar` width, shrink titlebar height as per
- update` ot-connect` icon (viewBox has a lot of white space and made the icon substantially smaller than the others
- update page css top padding to match new fixed titlebar height

## review requests
Please click through app and PD screens and make sure the new sizes don' [... cut]
commit 4d460117 (2018-06-25): fix(components): Resize main navbar and title bar to match designs (#1757)

closes #1285
closed 2018-06-25 by Kadee80