Opentrons/opentrons #1294: Remove text field from 'serialize name' checkbox

howisthisnamenottakenyet, 2018-04-27:
As a user I would like to not be exposed to unnecessary complexity in this maze of forms

--

## Background
Currently "serialize name" has it's own text field that creates a name separate from the ingredient name. Have not seen this as needed. 

## Acceptance Criteria
-Remove the serialized name text field. Clicking serialize name will now generate names in the format "{Ingredientname} {#}" Instead of "{Serializedname} {#}".

## Design
Just remove the text field

umbhau, 2018-06-08:
@IanLondon @howisthisnamenottakenyet Looks like this one is all done?

howisthisnamenottakenyet, 2018-06-08:
@umbhau not yet: 
[image]
[image]

IanLondon, 2018-06-08:
I just hid the text field in another ingred form ticket PR. I also still need to clean up the artifacts in the code left over from 'serializeName' text field

commit 22408c49 (2018-08-02): fix(protocol-designer): fix serialized name in ingred list

- factor apart utils and write tests for sortWells
- use sortWells where missing
- remove vestigal serializedName code

Closes #1294
PR #2002 (2018-08-03): fix(protocol-designer): fix serialized name in ingred list
##overview

Closes #1294 -- fixes IngredientItem (items in Ingredient List) text, for "serialized" ingredients. Was "Sample 1", "Sample 2" and now uses the ingredient name instead of "Sample"

Also remove old scars code for `serializeName` which went away a month or 2 ago

## changelog

- factor utils into a dir with index.js
- write tests for sortWells util
- use sortWells where missing
- remove vestigal serializedName code

## review requests

- [ ] In Add Ingredient Modal, check the "Serialize" box and save the ingredient form. The  [... cut]
commit d19d29b9 (2018-08-03): fix(protocol-designer): fix serialized name in ingred list (#2002)

- factor apart utils
- refactor + write tests for sortWells fn
- use sortWells where missing
- remove vestigal serializedName code

Closes #1294
closed 2018-08-03 by IanLondon