Opentrons/opentrons #1307: Pipette.delay Issue

umbhau, 2018-05-01:
Pipette.delay function exhibiting erratic behavior.

## Expected Behavior
- User can load a protocol including a pipette.delay

## Observed Behavior
- Loading a protocol including a pipette.delay results in an error message and _the robot immediately beginning to execute the protocol steps following the pipette.delay command_ :man-shrugging:

## Steps to recreate
- TBD

IanLondon, 2018-05-11:
May be related, may be a separate issue: when you upload a protocol (via the run app) that has a long delay, you actually have to wait that amount of time to proceed to calibration.

In addition to `pipette.delay`, some people also use `time.sleep` in their protocols, in an earlier version of protolib in PL we monkey-patched it so we could parse these protocols without sleeping for 2 hours or whatever time between steps

closed 2018-08-29 by Laura-Danielle

IanLondon, 2018-08-29:
@Laura-Danielle is this fixed? Do you know what PR fixed it?