Opentrons/opentrons #1317: Usability: Show where user is in calibration sub nav

pantslakz, 2018-05-01:
As as a user I would like to know where I am in the calibration process in the sub nav.

## Background
Currently as user progresses through calibration - we do not show a selected state for the calibration step they are on in the sub nav:

[image]

## Acceptance Criteria
 - Show selected state for the pipette/labware they are on.

## Design
- TBD

Kadee80, 2018-05-01:
@pantslakz  We have a selected state for selected/connected robots on the `RobotSettings` page. Might we start off by copying that selected/active state?

pantslakz, 2018-05-03:
@Kadee80 yes!

PR #1418 (2018-05-11): fix(app): Highlight active calibration step in subnav
## overview

This PR closes #1317 by adding an active state to selected labware or instrument item.

[image]

## changelog

- Implement active state style from connection panel in setup panel

## review requests

Another tiny PR. Just confirm selected labware or pipette matches screenshot for active list item.
commit 31efe08e (2018-05-11): fix(app): Highlight active calibration step in subnav

- Implement active state style from connection panel in setup panel
- Closes #1317
closed 2018-05-11 by Kadee80