Opentrons/opentrons #1318: Usability: Show correct run time and start time

pantslakz, 2018-05-01:
As a user, I would like the run time to start at 00:00:00 and view the correct time for when my run started.

## Background
Currently clock jumps above 00:00:00 and start time does not match what's on computer.

## Acceptance Criteria
 - start run at 00:00:00
- show correct time for start time

## Design
n/a

mcous, 2018-05-01:
Start time comes in to the app via the command log in the RPC session. This is known to be inaccurate and needs additional research

mcous, 2018-05-11:
Fixed by the addition of `startTime` to the RPC API `Session` in #1426 and #1427

closed 2018-05-11 by mcous

rvinzent, 2018-08-13:
This is still an issue for me on robot server and app version `3.2.0`. It seems the system date is incorrectly set on the raspberry pi. Our system clock on the robot is set to `Sun Jul  1 17:48:50 UTC 2018` as of August 13 2018.