Opentrons/opentrons #1349: Logs: Make Robot Logs Accessible

umbhau, 2018-05-02:
As a robot user, I would like to be able to easily get robot logs out of my robot.

## Acceptance Criteria
- [ ] Add an accessor function / fetch endpoint to fetch logs from Robot
- Fetch should return all logs by default (no need for search params / filters)

pantslakz, 2018-06-12:
@umbhau will this require UI in app in near future?

mcous, 2018-06-28:
Here's a summary of my research / conversations with @btmorr and @Laura-Danielle:

* The canonical way of responding with files is one file per HTTP response (i.e. no `multipart` stuff; it's not well supported for responses in browsers)
* Static routes to each log file in NGINX is a possibility
    * Would likely require API fallbacks until a Resin push can happen
* User should end up with a single file (probably ZIP) for attaching to intercom
    * @btmorr expressed trepidation in doing this in the API
    * Client can handle it; API log retrieval will need to happen in the app-shell thread then
    * This is ok, since we'll probably want the option bundle app logs (handled by app-shell) into the archive, too

## spec

- [ ] API should have an endpoint that returns the endpoint(s) for the log files themselves
    - This seems like it could go in `/health`
- [ ] API should have an endpoint per log file that returns the text file in the body
    - Depending on the size of the log file, you may want to consider streaming the response
    - Since app-shell will handle zipping stuff up, **no need** to set `Content-Disposition` header

Laura-Danielle, 2018-06-28:
@mcous why is it necessary to have an endpoint that returns other endpoints?

mcous, 2018-06-28:
@Laura-Danielle technically: best practices for a RESTful interface. In a "true" RESTful API, all endpoints responses would include the paths to other endpoints, such that you could discover the entire API just by making requests. In practice this rarely happens, though.

In this case: in my mind log file names or whatever are an implementation detail that should be allowed to change at any time without an app update. Sort of similar to how we have a `/positions` endpoint that we use to get positions before sending them to `/robot/move` rather than hardcoding the positions in the app.

PR #1928 (2018-07-26): feat(api): Add endpoints to get robot logs
## overview
Closes #1349.

## changelog
- Adds endpoint paths for logs in /health endpoint
- Adds endpoint to grab serial.log
- Adds endpoint to grab api.log

## review requests
Additional implementations needed? More info on 400 status needed?
closed 2018-07-26 by Laura-Danielle